gutierrezps/ESP32_I2C_Slave #1: Not working with ESP32-ArduinoMega I2C Connection

Nauman3S, 2020-07-15:
I have been trying to get this library working in ESP32-Arduino Mega communication but no luck. The Arduino is sending few simple chars over I2C while the ESP32 should receive and display the chars but it is not working. I am using 0x04 as an address in both devices and ESP32 is acting as a master.
What could be the issue here?

gutierrezps, 2020-07-15:
Are you using WirePacker on Arduino Mega? Please share the source codes of both Mega and ESP32.

Nauman3S, 2020-07-16:
ESP32 Code
`#include <Arduino.h>
#include <Wire.h>
#include <WireSlave.h>

#define SDA_PIN 23
#define SCL_PIN 22
#define I2C_SLAVE_ADDR 0x04

void receiveEvent(int howMany);

void setup()
{
    Serial.begin(115200);
    Serial.println("CONNECTING");

    bool success = WireSlave.begin(SDA_PIN, SCL_PIN, I2C_SLAVE_ADDR);
    if (!success) {
        Serial.println("I2C slave init failed");
        while(1) delay(100);
    }

    WireSlave.onReceive(receiveEvent);
}

void loop()
{
    // the slave response time is directly related to how often
    // this update() method is called, so avoid using long delays
    // inside loop()
    WireSlave.update();

    // let I2C and other ESP32 peripherals interrupts work
    delay(1);
}

// function that executes whenever a complete and valid packet
// is received from master
// this function is registered as an event, see setup()
void receiveEvent(int howMany)
{
    while (1 < WireSlave.available()) // loop through all but the last byte
    {
        char c = WireSlave.read();  // receive byte as a character
        Serial.print(c);            // print the character
    }

    int x = WireSlave.read();   // receive byte as an integer
    Serial.println(x);          // print the integer
`

Arduino Mega Code
`
#include <Wire.h>

void setup()
{
  Wire.begin(); // join i2c bus (address optional for master)
}

byte x = 0;

void loop()
{
  Wire.beginTransmission(0x04); // transmit to device #4
  Wire.write("x is ");        // sends five bytes
  Wire.write(x);              // sends one byte  
  Wire.endTransmission();    // stop transmitting

  x++;
  delay(500);
}`

Arduino IDE 1.8.12 is being used.

It gives an error when #include <WireSlave.h> is used on Arduino Mega Side:

"fatal error: driver/i2c.h: No such file or directory
 #include <driver/i2c.h>
          ^~~~~~~~~~~~~~"

gutierrezps, 2020-07-26:
I commited a fix to the master branch. Please download it, test it, and report if it worked.

MkLHX, 2020-09-17:
There is a way to send data without using WirePacker class?

I'm trying to send i2c data from raspberry pi to esp32 and nothing works...

Using arduino mini pro as master and esp32 as slave nothing works...

Then use snippets from this issue between arduino mini pro and esp32 and there is something transmit between 2 devices.

It's look the receiver need to get data formatted by WirePacker class. So how to deal with raspberry pi ?

Thx for your work because that a big shitty ESP32 don't have native i2c slave mode since 3-4 years....

Opus883, 2020-09-17:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}                                                                                                                                        The ESP32 does support native I2C slave interface, but this was not ported to the Arduino library. The ESP32_I2C_Slave implementation is a very good solution and solves the problem BUT it requires the use of WirePacker on the master and the slave. Therefore the RPi I2C implementation also needs the use of WirePacker to wrap the date before it is sent resp. unwrap to receive. If you are using a C++ tool chain for the RPi it should be possible to compile WirePacker. I will soon have a similar challenge when I want to use an ESP32 as an I2C slave for to an STM32. Good luck                                                                                                                                                                          Von: notifications@github.comGesendet: 17. September 2020 19:01An: ESP32_I2C_Slave@noreply.github.comAntworten: reply@reply.github.comCc: subscribed@noreply.github.comBetreff: Re: [gutierrezps/ESP32_I2C_Slave] Not working with ESP32-ArduinoMega I2C Connection (#1)  
There is a way to send data without using WirePacker class?
I'm trying to send i2c data from raspberry pi to esp32 and nothing works...
Using arduino mini pro as master and esp32 as slave nothing works...
Then use snippets from this issue between arduino mini pro and esp32 and there is something transmit between 2 devices.
It's look the receiver need to get data formatted by WirePacker class. So how to deal with raspberry pi ?
Thx for your work because that a big shitty ESP32 don't have native i2c slave mode since 3-4 years....

MkLHX, 2020-09-18:
@Opus883 thx

I know about the story of i2c slave mode on ESP32. I working on RPI with python, i'm going to work on the WirePaker alternative in python... I can see there is some work but nothing so hard ;-)

MkLHX, 2020-10-01:
you can find here a python library for Raspberry pi use
https://github.com/MkLHX/Raspberry_Pi_Master_for_ESP32_I2C_SLAVE

gutierrezps, 2021-02-21:
Tested with Arduino Nano as a master and ESP32 as slave, both writer-receiver and reader-sender, and they work.

closed 2021-02-21 by gutierrezps