gutierrezps/ESP32_I2C_Slave #2: Not working on ESP32 board MH-ET LIVE MINIKIT

MkLHX, 2020-09-17:
Hi there!

I'm trying to use this library on ESP32 has i2c slave with a raspberry pi 3A+ has master.
I'm working on platformio (core: 5.0.1 / Home 3.3.0) framework arduino
here my platformio.ini:
```
[env:mhetesp32minikit]
platform = espressif32
board = mhetesp32minikit
framework = arduino
monitor_speed = 115200
upload_speed = 921600

[env]
lib_deps =
  # Using a library name
  ArduinoUniqueID
  Board Identify
  Adafruit NeoPixel
  ESP32 I2C Slave
```

Bus connectivity is ok i can see ESP32 slave address when rpi is scanning. (tested with i2cdetect -y 1 and python I2C() =>scan() method).

But nothing happens when MASTER send i2c orders to the SLAVE.

My code is your examples code mixed.

```
#include <Arduino.h>
#include <Wire.h>
#include <WireSlave.h>

#define SDA_PIN 21
#define SCL_PIN 22
#define I2C_SLAVE_ADDR 0x14

void receiveEvent(int howMany);
void requestEvent();

void setup()
{
    Serial.begin(115200);

    bool success = WireSlave1.begin(SDA_PIN, SCL_PIN, I2C_SLAVE_ADDR);
    if (!success) {
        Serial.println("I2C slave init failed");
        while(1) delay(100);
    }

    WireSlave1.onReceive(receiveEvent);
    WireSlave1.onRequest(requestEvent);
}

void loop()
{
    // the slave response time is directly related to how often
    // this update() method is called, so avoid using long delays
    // inside loop()
    WireSlave1.update();

    // let I2C and other ESP32 peripherals interrupts work
    delay(1);
}

// function that executes whenever a complete and valid packet
// is received from master
// this function is registered as an event, see setup()
void receiveEvent(int howMany)
{
    while (1 < WireSlave1.available()) // loop through all but the last byte
    {
        char c = WireSlave1.read();  // receive byte as a character
        Serial.print(c);            // print the character
    }

    int x = WireSlave1.read();   // receive byte as an integer
    Serial.println(x);          // print the integer
}

void requestEvent()
{
    static byte y = 0;
    
    WireSlave1.print("y is ");
    WireSlave1.write(y++);
}

```
 
My python script:
```
import time
from adafruit_extended_bus import ExtendedI2C as I2C
from Adafruit_PureIO import smbus

# scan i2c devices
i2c = I2C(bus_id=1)
print(i2c.scan())

address = 0x14
with smbus.SMBus(1) as smbus:
    for i in range(0x00, 0x0A):
        print("register read: %s" % i)
        r = smbus.read_byte_data(address, i)
        print(hex(r))
        time.sleep(.3)

```

So what can i do to help? i need use esp32 has i2c slave ;-)

MkLHX, 2020-09-17:
FYI
I test examples code master r/w and slave r/w on a couple arduino pro mini (master) and esp32 MH-Et Live minikit (slave)

Same behavior...

gutierrezps, 2020-09-17:
On the master side, you need to use WirePacker to send data and WireSlaveRequest to receive data. Unfortunately I don't have a RPi, so I can't port those classes to Python.

closed 2020-09-28 by gutierrezps