gutierrezps/ESP32_I2C_Slave #3: Work on python master class

MkLHX, 2020-09-18:
Hello, i'm working on python class to permit Raspberry as i2c Master with ESP32 slave. look #2 

I have to convert the parcker and unpacker class into python code and i'm facing about this operation =>
https://github.com/gutierrezps/ESP32_I2C_Slave/blob/1224da32dd54f31ce62016438a25871d8d67324c/src/WirePacker.cpp#L63

What is the goal of byte_array + 2 ? and Why using packer and unpacker data workflow? is it the esp-idf layer who need this data format?

gutierrezps, 2020-09-18:
`buffer_` holds the following content (shown on README):

|            | Start | Length | Data\[0\] | Data\[1\] | ... | Data\[n-1\] |         CRC8        |  End |
|-----------:|:-----:|:------:|:---------:|:---------:|:---:|:-----------:|:-------------------:|:----:|
| byte index |   0   |    1   |     2     |     3     | ... |     n+1     |         n+2         |  n+3 |
|      value |  0x02 |   n+4  | data\[0\] | data\[1\] | ... | data\[n-1\] | crc of \[1..(n+1)\] | 0x04 |

CRC is calculated from `buffer_[1]` to `buffer_[n+1]`. So `buffer_[1]` which holds `totalLength` is added to the CRC on the previous line

https://github.com/gutierrezps/ESP32_I2C_Slave/blob/1224da32dd54f31ce62016438a25871d8d67324c/src/WirePacker.cpp#L62

and

https://github.com/gutierrezps/ESP32_I2C_Slave/blob/1224da32dd54f31ce62016438a25871d8d67324c/src/WirePacker.cpp#L63

updates the CRC with the contents of `data` i.e. `buffer[2...n+1]`.

gutierrezps, 2020-09-18:
> Why using packer and unpacker data workflow?

It's explained here: https://github.com/gutierrezps/ESP32_I2C_Slave#how-it-works

MkLHX, 2020-09-21:
@gutierrezps yeah i read and understand everything. but my question is really what is the goal in c++ side to make buffer_ + 2 operation?
buffer_ is an byte array and wat do the + 2 operation on this array? i cannot understand by my self sorry.

gutierrezps, 2020-09-21:
buffer_ is a pointer to the first byte of the array. buffer_ + 2 is therefore a pointer to the third byte of the array.

https://www.tutorialspoint.com/cplusplus/cpp_pointer_arithmatic.htm

MkLHX, 2020-09-23:
ok thx 👍

MkLHX, 2020-09-28:
Hello, here my work to allow raspberry pi to communicate with ESP32 as slave on i2c bus 
https://github.com/MkLHX/Raspberry_Pi_Master_for_ESP32_I2C_SLAVE

gutierrezps, 2020-09-28:
Great! I've added the link to your library on README.md.

closed 2020-09-28 by gutierrezps