gutierrezps/ESP32_I2C_Slave #7: slave esp32 not receiving anything in i2c Arduino wire library example

amin485, 2020-11-23:
Hi folks,
I am trying to use this library to communicate between two esp32, one as master and one as slave, particularly master needs to write to slave here. I have verified the following items, but still no receive event is detected by the code, and the function that is assigned by "WireSlave.onReceive(receiveEvent);" does not get called at all.
I appreciate any help on this

1. I checked where the thread travels within the WireSlave.cpp code once the "update" function is called, and the result always goes through the condition of "nothing received" or "inputLen == 0".

2. I have verified with Saleae logic hardware and probing the i2c signal and it shows that master successfully writes the bytes, and also slave acknowledges the receipt of write setup and each byte of packet.

3. I used the unpacker in master when sending

4. This print never shows up, meaning that nothing is detected to be received by the code
void receiveEvent(int howMany)
{
  #ifdef debug_prints
              Serial.print(" receiveEvent ");
#endif
}
[image]

gutierrezps, 2020-11-23:
Please provide a [minimal, reproducible example](https://stackoverflow.com/help/minimal-reproducible-example), with codes used on Master and Slave devices.

amin485, 2020-11-23:
Hi Gabriel,
The issue got resolved in a very weird way. I tried to reproduce it but i
could not do it until now,and i did not find a solid cause.
Among the things I suspect that can have a possibility of affecting are:
1- I switched from vscode to arduino (and ran the library basic example),
it worked fine, I tried to compile my own code in Arduino, it gave me some
compile errors, and then I got back to vscode compiling my own code, and it
got resolved right after.
2- at the time of problem and even right now after it got resolve, there
was an error regarding #include "driver/i2c.h":
cannot open source file "driver/i2c.h". but it works fine now regardless of
this problem.

If it happened again, I will let you know with more details.
thanks
Amin

gutierrezps, 2021-02-21:
Closing due to inactivity.

closed 2021-02-21 by gutierrezps