gutierrezps/ESP32_I2C_Slave #8: Error trying to send transmit string of 16 characters

targetarchitecture, 2021-01-18:
Hi Gabriel, 
  I'm using your library in my BBC Microbit & ESP32 board. I took your advice and have the BBC Microbit signalling a pin on the ESP32 to act as an interrupt. Everything is working well with sending text smaller than 16 characters in length. So my string 
"G,192.168.1.170" works fine. However if the string is longer such as "G1,192.168.1.170", then the following error occurs.

**sent: G1,192.168.1.170
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c", line 267, function: heap_caps_free
abort() was called at PC 0x40103957 on core 1

Backtrace: 0x40093a90:0x3ffdf8c0 0x40093cc1:0x3ffdf8e0 0x40103957:0x3ffdf900 0x4008593f:0x3ffdf930 0x400867a9:0x3ffdf950 0x4000bec7:0x3ffdf970 0x40138b3d:0x3ffdf990 0x4013902e:0x3ffdf9b0 0x400d2ff7:0x3ffdf9d0 0x400d66d9:0x3ffdfa20 0x400d2e3f:0x3ffdfad0 0x4008ff0d:0x3ffdfb00

Rebooting...
ets Jun  8 2016 00:22:57**

Any thoughts on how to resolve this?

Thanks John in York (UK)

targetarchitecture, 2021-01-18:
User error

closed 2021-01-18 by targetarchitecture

gutierrezps, 2021-01-19:
No problem, I'm glad you figured it out.

> I took your advice and have the BBC Microbit signalling a pin on the ESP32 to act as an interrupt

I had that in mind when I coded this library, but I couldn't find where I've written it. Are you sure I said this somewhere? 😅

targetarchitecture, 2021-01-19:
LOL, I think it was in an old forum that you posted to before you wrote this library.