gutierrezps/ESP32_I2C_Slave #13: ESP32 Wroom 32 as a slave

amankumbhani, 2021-05-07:
Hi,

I am trying to set up a network where an ESP32 development board will be the master & there will be 2 ESP32 slaves (dev modules) connected via SDA & SCL. 

I uploaded the master_reveiver to the ESP32 dev board & slave_sender to the ESP32 dev module. Somehow, the master is not able to identify slave & keeps printing "slave not found". 

Now I tried the same code, I uploaded the master_receiver to one ESP32 dev module & slave_sender to one ESP32 dev module, it worked. Is there any reason as to why the master ESP32 dev kit would not detect the slave?

Set the SLA pin as 23 & SDA as 22. Pull-up resistors connected on both slave boards (I am assuming the main esp32 dev kit doesn't require additional ones?)

Thanks! Any help would be appreciated!

gutierrezps, 2021-05-07:
On your first setup, have you assigned different I2C addresses to the ESP32 Slave modules? Run an I2C Scanner code and see if they're begin detected.

Add pull-ups on your setup with the ESP dev board. You just need one pair of pullups for the I2C bus, i.e. there's no need for pullups for each slave.

amankumbhani, 2021-05-08:
I have set the slave address at 0x04 for one slave module. When I run the code, it says 'slave successfully initiated at #4 address'. But when I run an I2C scanner, it shows there is a device at address 0x50. 

I even tried connecting to the detected address from the main master ESP32, still says the same.

amankumbhani, 2021-05-08:
[image]

This is the circuit which isn't working. The main esp is a development board - esp32 wroom 32. 

[image]

This is the circuit which is working properly. In this, both esp32s are dev module chips only.

gutierrezps, 2021-10-19:
Unfortunately, I couldn't address this issue in a timely manner because I haven't been working with the ESP32 ever since. In this meantime, I2C Slave support has been added to the official ESP32 framework as you can see in the updated README of this repo. Therefore this library is no longer needed and its development will be ceased.